adafruit/Adafruit_ImageReader #1: FeatherWingHX8357 Compile problem

ronguest, 2019-03-05:
At least for me the combination of Atom and PlatformIO is really productive and I’ve used it on a number of projects with Adafruit boards and libraries. However I’ve hit a wall on the ImageReader library. When I compile the example I get the error 

> error: no matching function for call to 'Adafruit_ImageReader::drawBMP(const char [13], Adafruit_HX8357&, int, int)'

Not surprisingly the example compiles fine if I use the Arduino IDE. Is there a workaround for this?

ladyada, 2019-03-05:
make sure you have the latest GFX library and all other libraries updated right - many updates hvae gone in. it sounds like you may have the wrong library versions somehwere

ronguest, 2019-03-05:
In the past I relied on PlatformIO’s notification about library updates and it wasn’t showing any waiting. But it looks like this was indeed the issue. I checked the Arduino IDE’s versions against PlatformIO’s and both GFX and the HX8357 libs were out of date. Once I manually updated them it compiled just fine. There were several other library updates as well. 

Thanks very much for the quick response. Feeling silly right now!

closed 2019-03-05 by ronguest