adafruit/Adafruit_ImageReader #2: ImageRead for OLED and TFT

marcwolf1960, 2019-03-26:
I have seen a lot of different modules - some cater only for OLED, others cater only for TFT.
I'm trying to help some people with a simple project so they can swap between TFT or OLED depending on cost.

If you could include an example for a SSD1331 that would be really great. 
Marc

ladyada, 2019-04-02:
should be basically the same as
https://github.com/adafruit/Adafruit_ImageReader/tree/master/examples
do you want to try to make an eample and submit it as a PR :)

ladyada, 2019-04-02:
@makermelissa you can take a look at this too!

marcwolf1960, 2019-04-03:
Hi.. I will be programming using Arduino code only.. Not Python.

I guess what I am trying to do is this

#IF TFT THEN
   TFT_Lib Display;
#ELSE
  OLED_Lib Display;
#END IF

Display.Clear;
Display.Draw_BMP( .....)

So the calling code is the same however the libraries will change. 
Dave

ladyada, 2019-04-03:
hiya no need, just clone of of the examples https://github.com/adafruit/Adafruit_ImageReader
replace it with the OLED display and submit it. its arduino only in this repo :)

PR #3 (2019-04-06): Added SSD1351 Examples
Fixes #2. I created one for the 1.5" and one for the 1.27".
closed 2019-04-06 by ladyada