adafruit/Adafruit_ImageReader #5: I can not stop to ImageReader.

OslafEman, 2019-04-12:
Hi, I started using the Adafruit ImageReader with a 1.4" TFT display and an ESP32.
My project includes also an keyboard and I'm trying to have the images sliding until any key got pressed in order to execute the rest of the code allowing other functions to use the TFT display..
But it does not happen, the images keep appearing.  I tried whit If() and while() statements but once the presentation starts no other functions can use the display.
I really appreciate your advice.  Thanks.
I attach a brief code with this behavior.

`#include <SPI.h>
#include <SD.h>
#include <Wire.h>
#include <i2ckeypad.h>
#include <Adafruit_GFX.h>    // Core graphics library
#include <Adafruit_ST7735.h> // Hardware-specific library
#include <Adafruit_ImageReader.h> // Image-reading functions
#include <RTClib.h>               // incluye libreria para el manejo del modulo RTC

#define ROWS 4
#define COLS 3
#define PCF8574_ADDR 0x20
i2ckeypad Teclado = i2ckeypad(PCF8574_ADDR, ROWS, COLS);

#define SD_CS    16 // SD card select pin

#define TFT_CS     26
#define TFT_RST    -1
#define TFT_DC     5
Adafruit_ST7735 tft = Adafruit_ST7735(TFT_CS,  TFT_DC, TFT_RST);

Adafruit_ImageReader reader;     // Class w/image-reading functions
Adafruit_Image       img;        // An image loaded into RAM
int32_t              width  = 0, // BMP image dimensions
                     height = 0;

char Text[16];
int Ind = 0;

void setup() {
  Serial.begin(115200);

  ImageReturnCode stat; // Status from image-reading functions

  tft.initR(INITR_144GREENTAB); // initialise ST7735S chip, green tab
  tft.setTextWrap(true); // Allow text to run off right edge
  tft.fillScreen(ST7735_WHITE);
  tft.setTextColor(ST7735_BLACK);

  Serial.print(F("Initializing SD card..."));
  if (!SD.begin(SD_CS)) {
    Serial.println(F("failed!"));
    for (;;); // Loop here forever
  }
  Serial.println(F("OK!"));

  Wire.begin();
  Teclado.init();

  Serial.print("Testing keypad/PCF8574 I2C port expander arduino lib\n\n");
}

void loop()
{
  char Key = Teclado.get_key();

  if (Key) {
    Text[Ind] = Key;
    Ind++;
    tft.setCursor(1, 36);
    tft.println(Text);
  }
  reader.drawBMP("/image1.bmp", tft, 0, 0);
  delay(3000);
  reader.drawBMP("/image2.bmp", tft, 0, 0);
  delay(3000);
  reader.drawBMP("/image3.bmp", tft, 0, 0);
  delay(3000);
}`

ladyada, 2019-08-07:
hiya this isn't an issue with the library, but in how to code for arduino. please contact the place you purchased your display/arduino for tech support :)

closed 2019-08-07 by ladyada

OslafEman, 2019-08-07:
Thanks ladyada.  You're right.  I've already solved this issue :)