adafruit/Adafruit_ImageReader #26: Error compiling for board Arduino Nano 33 BLE.

bergstar, 2020-02-06:
- Arduino board:  Arduino Nano 33 BLE

- Arduino IDE version (found in Arduino -> About Arduino menu):  1.8.10

- List the steps to reproduce the problem below (if possible attach a sketch or
  copy the sketch code in too): 
Select target board and try to compile:
```
#include <Adafruit_ImageReader.h>
```
Following error dump will be given:
```
Arduino: 1.8.10 (Windows 10), Board: "Arduino Nano 33 BLE"

In file included from c:\users\arduino\appdata\local\arduino15\packages\arduino\tools\arm-none-eabi-gcc\7-2017q4\arm-none-eabi\include\sys\_default_fcntl.h:200:0,

                 from c:\users\arduino\appdata\local\arduino15\packages\arduino\tools\arm-none-eabi-gcc\7-2017q4\arm-none-eabi\include\sys\fcntl.h:4,

                 from c:\users\arduino\appdata\local\arduino15\packages\arduino\tools\arm-none-eabi-gcc\7-2017q4\arm-none-eabi\include\fcntl.h:1,

                 from c:\users\arduino\Documents\Arduino\libraries\SdFat_-_Adafruit_Fork\src/FatLib/FatApiConstants.h:30,

                 from c:\users\arduino\Documents\Arduino\libraries\SdFat_-_Adafruit_Fork\src/FatLib/FatFile.h:36,

                 from c:\users\arduino\Documents\Arduino\libraries\SdFat_-_Adafruit_Fork\src/FatLib/ArduinoFiles.h:33,

                 from c:\users\arduino\Documents\Arduino\libraries\SdFat_-_Adafruit_Fork\src/FatLib/FatLib.h:27,

                 from c:\users\arduino\Documents\Arduino\libraries\SdFat_-_Adafruit_Fork\src/SdFat.h:33,

                 from c:\users\arduino\Documents\Arduino\libraries\Adafruit_SPIFlash\src/Adafruit_SPIFlash.h:35,

                 from c:\users\arduino\Documents\Arduino\libraries\Adafruit_ImageReader_Library/Adafruit_ImageReader.h:18,

                 from c:\users\arduino\AppData\Local\Temp\arduino_modified_sketch_897396\BreakoutSSD1351.ino:2:

c:\users\arduino\appdata\local\arduino15\packages\arduino\tools\arm-none-eabi-gcc\7-2017q4\arm-none-eabi\include\sys\stat.h:27:8: error: redefinition of 'struct stat'

 struct stat

        ^~~~

In file included from c:\users\arduino\AppData\Local\Arduino15\packages\arduino\hardware\mbed\1.1.4\cores\arduino/mbed/platform/mbed_error.h:21:0,

                 from c:\users\arduino\AppData\Local\Arduino15\packages\arduino\hardware\mbed\1.1.4\cores\arduino/mbed/rtos/Queue.h:28,

                 from c:\users\arduino\AppData\Local\Arduino15\packages\arduino\hardware\mbed\1.1.4\cores\arduino/mbed/rtos/Mail.h:28,

                 from c:\users\arduino\AppData\Local\Arduino15\packages\arduino\hardware\mbed\1.1.4\cores\arduino/mbed/rtos/rtos.h:33,

                 from c:\users\arduino\AppData\Local\Arduino15\packages\arduino\hardware\mbed\1.1.4\cores\arduino/mbed/mbed.h:23,

                 from c:\users\arduino\AppData\Local\Arduino15\packages\arduino\hardware\mbed\1.1.4\cores\arduino/Arduino.h:32,

                 from sketch\BreakoutSSD1351.ino.cpp:1:

c:\users\arduino\AppData\Local\Arduino15\packages\arduino\hardware\mbed\1.1.4\cores\arduino/mbed/platform/mbed_retarget.h:500:8: note: previous definition of 'struct stat'

 struct stat {

        ^~~~

Multiple libraries were found for "Adafruit_SPITFT.h"
 Used: c:\users\arduino\Documents\Arduino\libraries\Adafruit_GFX_Library
Multiple libraries were found for "Adafruit_EPD.h"
 Used: c:\users\arduino\Documents\Arduino\libraries\Adafruit_EPD
Multiple libraries were found for "Adafruit_ImageReader.h"
 Used: c:\users\arduino\Documents\Arduino\libraries\Adafruit_ImageReader_Library
Multiple libraries were found for "Adafruit_SPIFlash.h"
 Used: c:\users\arduino\Documents\Arduino\libraries\Adafruit_SPIFlash
Multiple libraries were found for "SPI.h"
 Used: c:\users\arduino\AppData\Local\Arduino15\packages\arduino\hardware\mbed\1.1.4\libraries\SPI
Multiple libraries were found for "SdFat.h"
 Used: c:\users\arduino\Documents\Arduino\libraries\SdFat_-_Adafruit_Fork
exit status 1
Error compiling for board Arduino Nano 33 BLE.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
```

Same error has been reported here:
https://forum.arduino.cc/index.php?topic=647949.0

ladyada, 2020-02-06:
hi it does not seem like theres much we can do about it, the errors are in the underlying Arduino cores

bergstar, 2020-02-07:
@ladyada thank you for reply. I'm trying to report it to Android support. Will follow up this issue when its resolved. I think it can be good to keep it open for other people to be aware of the issue.

SpenceKonde, 2020-02-07:
It looks to me like the issue is that the library is naming a struct the same thing ('stat') that the core library uses for a different purpose? If my analysis is correct, that is an issue with the library - a third party library should not use a name that conflicts with the core libraries.  And fixing it would just be a find/replace job...

ladyada, 2020-02-07:
the error is flagging 

c:\users\arduino\appdata\local\arduino15\packages\arduino\tools\arm-none-eabi-gcc\7-2017q4\arm-none-eabi\include\sys\stat.h

c:\users\arduino\AppData\Local\Arduino15\packages\arduino\hardware\mbed\1.1.4\cores\arduino/mbed/platform/mbed_retarget.h:500:8: note: previous definition of 'struct stat'

these are not 3rd party library files, they're core files

SpenceKonde, 2020-02-07:
Ah, I see. So this is a bug in the Arduino core, and should be reported to the Arduino folks.

Tofer89, 2020-05-22:
Did anyone find a workarround to make the ImageReader compile on Arduino Nano 33 BLE?

I tried to modify the `SdfatConfig.h` as discribed here (https://github.com/greiman/SdFat-beta/issues/35) and I exchanged the latest `mbed_retarget.h` as described here:

https://github.com/arduino/ArduinoCore-nRF528x-mbedos/issues/49
https://github.com/ARMmbed/mbed-os/pull/12607/files

Both solutions help to fix the Sdfat lib to compile. But I'm still struggeling to compile ImageReader.