adafruit/Adafruit_ImageReader #27: Failed to compile on Arduino Nano Every

Tofer89, 2020-04-30:
- Arduino board:  **Arduino Nano Every**

- Arduino IDE version (found in Arduino -> About Arduino menu):  **1.8.12**

- List the steps to reproduce the problem below (if possible attach a sketch or
  copy the sketch code in too): **#include <Adafruit_ImageReader.h>**

**Error:**
`Arduino:` 1.8.12 (Windows 10), Board: "Arduino Nano Every, ATMEGA328"

In file included from C:\Users\Christofer\Documents\Arduino\libraries\SdFat_-_Adafruit_Fork\src/SdCard/SdSpiCard.h:35:0,

                 from C:\Users\Christofer\Documents\Arduino\libraries\SdFat_-_Adafruit_Fork\src/BlockDriver.h:32,

                 from C:\Users\Christofer\Documents\Arduino\libraries\SdFat_-_Adafruit_Fork\src/SdFat.h:32,

                 from C:\Users\Christofer\Desktop\Arduino\Einzelbeispiele\Test\Test.ino:2:

c:\users\christofer\documents\arduino\libraries\sdfat_-_adafruit_fork\src\spidriver\sdspidriver.h: In member function 'uint8_t SdSpiAltDriver::receive()':

c:\users\christofer\documents\arduino\libraries\sdfat_-_adafruit_fork\src\spidriver\sdspidriver.h:392:3: error: 'SPDR' was not declared in this scope

   SPDR = 0XFF;

   ^~~~

c:\users\christofer\documents\arduino\libraries\sdfat_-_adafruit_fork\src\spidriver\sdspidriver.h:392:3: note: suggested alternative: 'SPI'

   SPDR = 0XFF;

   ^~~~

   SPI

c:\users\christofer\documents\arduino\libraries\sdfat_-_adafruit_fork\src\spidriver\sdspidriver.h:393:12: error: 'SPSR' was not declared in this scope

   while (!(SPSR & (1 << SPIF))) {}

            ^~~~

c:\users\christofer\documents\arduino\libraries\sdfat_-_adafruit_fork\src\spidriver\sdspidriver.h:393:12: note: suggested alternative: 'SPI'

   while (!(SPSR & (1 << SPIF))) {}

            ^~~~

            SPI

c:\users\christofer\documents\arduino\libraries\sdfat_-_adafruit_fork\src\spidriver\sdspidriver.h:393:25: error: 'SPIF' was not declared in this scope

   while (!(SPSR & (1 << SPIF))) {}

                         ^~~~

c:\users\christofer\documents\arduino\libraries\sdfat_-_adafruit_fork\src\spidriver\sdspidriver.h:393:25: note: suggested alternative: 'SPI'

   while (!(SPSR & (1 << SPIF))) {}

                         ^~~~

                         SPI

c:\users\christofer\documents\arduino\libraries\sdfat_-_adafruit_fork\src\spidriver\sdspidriver.h: In member function 'uint8_t SdSpiAltDriver::receive(uint8_t*, size_t)':

c:\users\christofer\documents\arduino\libraries\sdfat_-_adafruit_fork\src\spidriver\sdspidriver.h:401:3: error: 'SPDR' was not declared in this scope

   SPDR = 0XFF;

   ^~~~

c:\users\christofer\documents\arduino\libraries\sdfat_-_adafruit_fork\src\spidriver\sdspidriver.h:401:3: note: suggested alternative: 'SPI'

   SPDR = 0XFF;

   ^~~~

   SPI

c:\users\christofer\documents\arduino\libraries\sdfat_-_adafruit_fork\src\spidriver\sdspidriver.h:403:14: error: 'SPSR' was not declared in this scope

     while (!(SPSR & (1 << SPIF))) {}

              ^~~~

c:\users\christofer\documents\arduino\libraries\sdfat_-_adafruit_fork\src\spidriver\sdspidriver.h:403:14: note: suggested alternative: 'SPI'

     while (!(SPSR & (1 << SPIF))) {}

              ^~~~

              SPI

c:\users\christofer\documents\arduino\libraries\sdfat_-_adafruit_fork\src\spidriver\sdspidriver.h:403:27: error: 'SPIF' was not declared in this scope

     while (!(SPSR & (1 << SPIF))) {}

                           ^~~~

c:\users\christofer\documents\arduino\libraries\sdfat_-_adafruit_fork\src\spidriver\sdspidriver.h:403:27: note: suggested alternative: 'SPI'

     while (!(SPSR & (1 << SPIF))) {}

                           ^~~~

                           SPI

c:\users\christofer\documents\arduino\libraries\sdfat_-_adafruit_fork\src\spidriver\sdspidriver.h:408:12: error: 'SPSR' was not declared in this scope

   while (!(SPSR & (1 << SPIF))) {}

            ^~~~

c:\users\christofer\documents\arduino\libraries\sdfat_-_adafruit_fork\src\spidriver\sdspidriver.h:408:12: note: suggested alternative: 'SPI'

   while (!(SPSR & (1 << SPIF))) {}

            ^~~~

            SPI

c:\users\christofer\documents\arduino\libraries\sdfat_-_adafruit_fork\src\spidriver\sdspidriver.h:408:25: error: 'SPIF' was not declared in this scope

   while (!(SPSR & (1 << SPIF))) {}

                         ^~~~

c:\users\christofer\documents\arduino\libraries\sdfat_-_adafruit_fork\src\spidriver\sdspidriver.h:408:25: note: suggested alternative: 'SPI'

   while (!(SPSR & (1 << SPIF))) {}

                         ^~~~

                         SPI

c:\users\christofer\documents\arduino\libraries\sdfat_-_adafruit_fork\src\spidriver\sdspidriver.h: In member function 'void SdSpiAltDriver::send(uint8_t)':

c:\users\christofer\documents\arduino\libraries\sdfat_-_adafruit_fork\src\spidriver\sdspidriver.h:414:3: error: 'SPDR' was not declared in this scope

   SPDR = data;

   ^~~~

c:\users\christofer\documents\arduino\libraries\sdfat_-_adafruit_fork\src\spidriver\sdspidriver.h:414:3: note: suggested alternative: 'SPI'

   SPDR = data;

   ^~~~

   SPI

c:\users\christofer\documents\arduino\libraries\sdfat_-_adafruit_fork\src\spidriver\sdspidriver.h:415:12: error: 'SPSR' was not declared in this scope

   while (!(SPSR & (1 << SPIF))) {}

            ^~~~

c:\users\christofer\documents\arduino\libraries\sdfat_-_adafruit_fork\src\spidriver\sdspidriver.h:415:12: note: suggested alternative: 'SPI'

   while (!(SPSR & (1 << SPIF))) {}

            ^~~~

            SPI

c:\users\christofer\documents\arduino\libraries\sdfat_-_adafruit_fork\src\spidriver\sdspidriver.h:415:25: error: 'SPIF' was not declared in this scope

   while (!(SPSR & (1 << SPIF))) {}

                         ^~~~

c:\users\christofer\documents\arduino\libraries\sdfat_-_adafruit_fork\src\spidriver\sdspidriver.h:415:25: note: suggested alternative: 'SPI'

   while (!(SPSR & (1 << SPIF))) {}

                         ^~~~

                         SPI

c:\users\christofer\documents\arduino\libraries\sdfat_-_adafruit_fork\src\spidriver\sdspidriver.h: In member function 'void SdSpiAltDriver::send(const uint8_t*, size_t)':

c:\users\christofer\documents\arduino\libraries\sdfat_-_adafruit_fork\src\spidriver\sdspidriver.h:422:3: error: 'SPDR' was not declared in this scope

   SPDR = buf[0];

   ^~~~

c:\users\christofer\documents\arduino\libraries\sdfat_-_adafruit_fork\src\spidriver\sdspidriver.h:422:3: note: suggested alternative: 'SPI'

   SPDR = buf[0];

   ^~~~

   SPI

c:\users\christofer\documents\arduino\libraries\sdfat_-_adafruit_fork\src\spidriver\sdspidriver.h:427:16: error: 'SPSR' was not declared in this scope

       while (!(SPSR & (1 << SPIF))) {}

                ^~~~

c:\users\christofer\documents\arduino\libraries\sdfat_-_adafruit_fork\src\spidriver\sdspidriver.h:427:16: note: suggested alternative: 'SPI'

       while (!(SPSR & (1 << SPIF))) {}

                ^~~~

                SPI

c:\users\christofer\documents\arduino\libraries\sdfat_-_adafruit_fork\src\spidriver\sdspidriver.h:427:29: error: 'SPIF' was not declared in this scope

       while (!(SPSR & (1 << SPIF))) {}

                             ^~~~

c:\users\christofer\documents\arduino\libraries\sdfat_-_adafruit_fork\src\spidriver\sdspidriver.h:427:29: note: suggested alternative: 'SPI'

       while (!(SPSR & (1 << SPIF))) {}

                             ^~~~

                             SPI

c:\users\christofer\documents\arduino\libraries\sdfat_-_adafruit_fork\src\spidriver\sdspidriver.h:435:12: error: 'SPSR' was not declared in this scope

   while (!(SPSR & (1 << SPIF))) {}

            ^~~~

c:\users\christofer\documents\arduino\libraries\sdfat_-_adafruit_fork\src\spidriver\sdspidriver.h:435:12: note: suggested alternative: 'SPI'

   while (!(SPSR & (1 << SPIF))) {}

            ^~~~

            SPI

c:\users\christofer\documents\arduino\libraries\sdfat_-_adafruit_fork\src\spidriver\sdspidriver.h:435:25: error: 'SPIF' was not declared in this scope

   while (!(SPSR & (1 << SPIF))) {}

                         ^~~~

c:\users\christofer\documents\arduino\libraries\sdfat_-_adafruit_fork\src\spidriver\sdspidriver.h:435:25: note: suggested alternative: 'SPI'

   while (!(SPSR & (1 << SPIF))) {}

                         ^~~~

                         SPI

exit status 1
Error compiling for board Arduino Nano Every.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> `Preferences.`

ladyada, 2020-04-30:
hi the nano is a significantly different chip, if you'd like to submit a working PR to sdfat library, we'd accept it but we have no hardware to test with so we cannot show this library working with nano

FusionOpz, 2020-12-22:
> hi the nano is a significantly different chip, if you'd like to submit a working PR to sdfat library, we'd accept it but we have no hardware to test with so we cannot show this library working with nano

When you change `#define SPI_DRIVER_SELECT 0` to `#define SPI_DRIVER_SELECT 1 ` on the sdfat library it works, just can't get imagereader to work with 2.0.2...