adafruit/Adafruit_ImageReader #37: Not necessarily bugs, but recommendations…

PaintYourDragon, 2020-10-28:
**First:** would suggest renaming one or two of the examples, specifically EInkFeatherWing and/or EInkGray29BmpButtonDemo. The grayscale FeatherWing is labeled “2.9" E-Ink FeatherWing” on the back, and running the EInkFeatherWing sketch here _almost sorta_ works, just enough that some guy who’s not the sharpest stick might’ve spent an entire day trying to figure out what’s wrong with his BMP, his Arduino code and/or delved deep into the ImageReader source looking for the “bug” when it was really just that he needed to use the other example that, while it does have “Gray” in the name, “Gray” doesn’t appear on the FeatherWing and “FeatherWing” does not appear in the example name. He doesn’t get out often.

**Second:** I don’t know if this is a bug/typo or was an intentional choice done for perceptual or legibility reasons, but noticed while I was in the code that on grayscale displays, brightness levels below 0x60 (rather than 0x40) are assigned black. The image I was testing with just happened to be a gray ramp, and with this threshold level the range of dark grays get “the squeeze.” Can’t load the original BMP here because Github, so here’s a JPG substitute:

[image]

And how it appears on the ’Wing:

[image]

But again, if that’s intentional…maybe it’s to make most line art look right or something…please disregard.

ladyada, 2020-10-28:
i think i made a hex typo for the level detection. can u post up the bmp of gray-diant?

PaintYourDragon, 2020-10-28:
Had to ZIP it because Github and file types…

[gray-ramp.bmp.zip](https://github.com/adafruit/Adafruit_ImageReader/files/5453551/gray-ramp.bmp.zip)

makermelissa, 2020-10-28:
I merged in a new ThinkInk demo which will eventually supplant both the EInkFeatherWing and EInkBreakouts demos.

tannewt, 2020-10-28:
Here is CircuitPython's ramp. We use a luma computation instead of a direct average. [image]

tannewt, 2020-10-28:
Note that we have slightly different settings at the moment that lead to a darker light gray.

ladyada, 2020-10-28:
@PaintYourDragon iz up to you - how do you want the grayscale on arduino to matchymatch?

tannewt, 2020-10-28:
Here is our code: https://github.com/adafruit/circuitpython/blob/main/shared-module/displayio/ColorConverter.c#L54-L59

I can't quite remember where I got that formula. Probably wikipedia.