adafruit/Adafruit_ImageReader #43: SD Card initialization failed

guljanjua, 2021-04-06:
Hi,

I hope you are doing good. I have this adafruit display: [https://www.adafruit.com/product/3533](url)
- Arduino board:  Arduino Nano
- Arduino IDE version 1.8.13

But this tutorial is working for me.

[](https://github.com/adafruit/Adafruit-ST7735-Library/tree/master/examples/graphicstest)

I am trying this tutorial and it is not working.

[](https://github.com/adafruit/Adafruit_ImageReader/blob/master/examples/BreakoutST7735-160x80/BreakoutST7735-160x80.ino)

It says SD card initialization failed.

However if I test it with SD library of Arduino this example works:
https://github.com/arduino-libraries/SD/blob/master/examples/listfiles/listfiles.ino

- So In the end I am sure that SD card is fine and it is working with Arduino SD library. 
- SPI communication is working fine as both SD card and Graphics tests are working fine. 

Please help.

caternuson, 2021-04-06:
Thanks! Adding forum post for ref:
https://forums.adafruit.com/viewtopic.php?f=47&t=177750

guljanjua, 2021-04-06:
Hi,

Yes that is my post on adafruit forum. Can you help me in solving this issue?