adafruit/Adafruit_ImageReader #45: Will not compile

orenzav, 2021-09-10:
Hi, 

I'm using the ImageReader example code (see below) to display images from an SD card. Hardware is an Adfruit ESP32 Feather and an Adafruit 1.44" Color TFT LCD Display with MicroSD Card breakout - ST7735R. I'm using PlatformIO (VScode).

No matter what I seem to try I get a compiler error which appears to suggest that **flash_t was not declared in scope for Adafruit_FlashTransport_ESP32.cpp**

Interestingly (maybe) when I try and compile using Arduino 1.8.15 IDE it works. 

Conversely if I try the same code again on Arduino 2.0 Beta11 IDE it also fails with the same error message above that PlatformIO gives me.

I've searched the forums buy closest I've come across that's similar is this: https://github.com/espressif/arduino-esp32/issues/4115 which doesn't work

**Error:**
c:\Users\user\Documents\Arduino\libraries\Adafruit_SPIFlash\src\esp32\Adafruit_FlashTransport_ESP32.cpp: In member function 'SPIFlash_Device_t* Adafruit_FlashTransport_ESP32::getFlashDevice()':
c:\Users\user\Documents\Arduino\libraries\Adafruit_SPIFlash\src\esp32\Adafruit_FlashTransport_ESP32.cpp:54:3: error: 'esp_flash_t' was not declared in this scope
   esp_flash_t const *flash = _partition->flash_chip;
   ^
c:\Users\user\Documents\Arduino\libraries\Adafruit_SPIFlash\src\esp32\Adafruit_FlashTransport_ESP32.cpp:55:36: error: 'flash' was not declared in this scope
   _flash_device.manufacturer_id = (flash->chip_id >> 16);
                                    ^
Compilation error: Error: 2 UNKNOWN: exit status 1

**Code:**
// Adafruit_ImageReader test for Adafruit ST7735 TFT Breakout for Arduino.
// Demonstrates loading images from SD card or flash memory to the screen,
// to RAM, and how to query image file dimensions.
// Requires three BMP files in root directory of SD card:
// rgbwheel.bmp, miniwoof.bmp and wales.bmp.
// As written, this uses the microcontroller's SPI interface for the screen
// (not 'bitbang') and must be wired to specific pins (e.g. for Arduino Uno,
// MOSI = pin 11, MISO = 12, SCK = 13). Other pins are configurable below.

#include <Adafruit_GFX.h>         // Core graphics library
#include <Adafruit_ST7735.h>      // Hardware-specific library
#include <SdFat.h>                // SD card & FAT filesystem library
#include <Adafruit_SPIFlash.h>    // SPI / QSPI flash library
#include <Adafruit_ImageReader.h> // Image-reading functions

// Comment out the next line to load from SPI/QSPI flash instead of SD card:
#define USE_SD_CARD

// TFT display and SD card share the hardware SPI interface, using
// 'select' pins for each to identify the active device on the bus.

#define SD_CS    33 // SD card select pin
#define TFT_CS   14 // TFT select pin
#define TFT_DC   32 // TFT display/command pin
#define TFT_RST  15 // Or set to -1 and connect to Arduino RESET pin

#if defined(USE_SD_CARD)
  SdFat                SD;         // SD card filesystem
  Adafruit_ImageReader reader(SD); // Image-reader object, pass in SD filesys
#else
  // SPI or QSPI flash filesystem (i.e. CIRCUITPY drive)
  #if defined(__SAMD51__) || defined(NRF52840_XXAA)
    Adafruit_FlashTransport_QSPI flashTransport(PIN_QSPI_SCK, PIN_QSPI_CS,
      PIN_QSPI_IO0, PIN_QSPI_IO1, PIN_QSPI_IO2, PIN_QSPI_IO3);
  #else
    #if (SPI_INTERFACES_COUNT == 1)
      Adafruit_FlashTransport_SPI flashTransport(SS, &SPI);
    #else
      Adafruit_FlashTransport_SPI flashTransport(SS1, &SPI1);
    #endif
  #endif
  Adafruit_SPIFlash    flash(&flashTransport);
  FatFileSystem        filesys;
  Adafruit_ImageReader reader(filesys); // Image-reader, pass in flash filesys
#endif

Adafruit_ST7735      tft    = Adafruit_ST7735(TFT_CS, TFT_DC, TFT_RST);
Adafruit_Image       img;        // An image loaded into RAM
int32_t              width  = 0, // BMP image dimensions
                     height = 0;

void setup(void) {

  ImageReturnCode stat; // Status from image-reading functions

  Serial.begin(9600);
#if !defined(ESP32)
  while(!Serial);       // Wait for Serial Monitor before continuing
#endif

  tft.initR(INITR_144GREENTAB); // Initialize screen

  // The Adafruit_ImageReader constructor call (above, before setup())
  // accepts an uninitialized SdFat or FatFileSystem object. This MUST
  // BE INITIALIZED before using any of the image reader functions!
  Serial.print(F("Initializing filesystem..."));
#if defined(USE_SD_CARD)
  // SD card is pretty straightforward, a single call...
  if(!SD.begin(SD_CS, SD_SCK_MHZ(10))) { // Breakouts require 10 MHz limit due to longer wires
    Serial.println(F("SD begin() failed"));
    for(;;); // Fatal error, do not continue
  }
#else
  // SPI or QSPI flash requires two steps, one to access the bare flash
  // memory itself, then the second to access the filesystem within...
  if(!flash.begin()) {
    Serial.println(F("flash begin() failed"));
    for(;;);
  }
  if(!filesys.begin(&flash)) {
    Serial.println(F("filesys begin() failed"));
    for(;;);
  }
#endif
  Serial.println(F("OK!"));

  // Fill screen blue. Not a required step, this just shows that we're
  // successfully communicating with the screen.
  tft.fillScreen(ST7735_BLUE);

  // Load full-screen BMP file 'lily128.bmp' at position (0,0) (top left).
  // Notice the 'reader' object performs this, with 'tft' as an argument.
  Serial.print(F("Loading lily128.bmp to screen..."));
  stat = reader.drawBMP("/lily128.bmp", tft, 0, 0);
  reader.printStatus(stat);   // How'd we do?
/*
  // Query the dimensions of image 'miniwoof.bmp' WITHOUT loading to screen:
  Serial.print(F("Querying miniwoof.bmp image size..."));
  stat = reader.bmpDimensions("/miniwoof.bmp", &width, &height);
  reader.printStatus(stat);   // How'd we do?
  if(stat == IMAGE_SUCCESS) { // If it worked, print image size...
    Serial.print(F("Image dimensions: "));
    Serial.print(width);
    Serial.write('x');
    Serial.println(height);
  }

  // Load small BMP 'wales.bmp' into a GFX canvas in RAM. This should fail
  // gracefully on Arduino Uno and other small devices, meaning the image
  // will not load, but this won't make the program stop or crash, it just
  // continues on without it. Should work on Arduino Mega, Zero, etc.
  Serial.print(F("Loading wales.bmp to canvas..."));
  stat = reader.loadBMP("/wales.bmp", img);
  reader.printStatus(stat); // How'd we do?
*/
  delay(5000); // Pause 5 seconds before moving on to loop()
  
}

void loop() {
  
}

hathach, 2021-09-11:
@orenzav I edited your post to make it easier to read, can you also provide the version of following libraries for reproducing:
- BSP of arduino-esp32
- Adafruit_ImageReader
- Adafruit_SPIFlash

also please attached your full compile log from Arduino IDE for analysis

orenzav, 2021-09-11:
Hi, 

Thanks for your reply:

Adafruit_ImageReader is version 2.6.2
Adafruit_SPIFlash  is version 3.7.0
BSP of arduino-esp32 - Really sorry not sure what this is where would I find it? Do you mean platform version as per below its Espressif 32 (3.3.2)

Full compile log from fresh as per below:

`Processing featheresp32 (platform: espressif32; board: featheresp32; framework: arduino)
-------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/featheresp32.html
PLATFORM: Espressif 32 (3.3.2) > Adafruit ESP32 Feather
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 3.10006.210326 (1.0.6)
 - tool-esptoolpy 1.30100.210531 (3.1.0)
 - tool-mkspiffs 2.230.0 (2.30)
 - toolchain-xtensa32 2.50200.97 (5.2.0)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Library Manager: Installing adafruit/Adafruit GFX Library @ ^1.10.10
Library Manager: Adafruit GFX Library @ 1.10.11 has been installed!
Library Manager: Installing dependencies...
Library Manager: Installing Adafruit BusIO
Library Manager: Warning! More than one package has been found by Adafruit BusIO requirements:
 - adafruit/Adafruit BusIO @ 1.9.1
 - mertmechanic/Adafruit BusIO @ 1.7.3
Library Manager: Please specify detailed REQUIREMENTS using package owner and version (showed above) to avoid name conflicts
Library Manager: Adafruit BusIO @ 1.9.1 has been installed!
Library Manager: Installing adafruit/Adafruit ST7735 and ST7789 Library @ ^1.7.3
Library Manager: Adafruit ST7735 and ST7789 Library @ 1.7.4 has been installed!
Library Manager: Installing dependencies...
Library Manager: Adafruit GFX Library @ 1.10.11 is already installed
Library Manager: Installing Adafruit seesaw Library
Library Manager: Adafruit seesaw Library @ 1.4.5 has been installed!
Library Manager: Installing dependencies...
Library Manager: Adafruit ST7735 and ST7789 Library @ 1.7.4 is already installed
Library Manager: Installing SD
Library Manager: Warning! More than one package has been found by SD requirements:
 - adafruit/SD @ 0.0.0-alpha+sha.041f788250
 - arduino-libraries/SD @ 1.2.4
Library Manager: Please specify detailed REQUIREMENTS using package owner and version (showed above) to avoid name conflicts
Library Manager: SD @ 0.0.0-alpha+sha.041f788250 has been installed!
Library Manager: Adafruit BusIO @ 1.9.1 is already installed
Library Manager: Installing adafruit/Adafruit ImageReader Library @ ^2.6.2
Library Manager: Adafruit ImageReader Library @ 2.6.2 has been installed!
Library Manager: Installing dependencies...
Library Manager: Adafruit GFX Library @ 1.10.11 is already installed
Library Manager: Adafruit ST7735 and ST7789 Library @ 1.7.4 is already installed
Library Manager: Installing Adafruit HX8357 Library
Library Manager: Adafruit HX8357 Library @ 1.1.11 has been installed!
Library Manager: Installing dependencies...
Library Manager: Installing Adafruit STMPE610
Library Manager: Adafruit STMPE610 @ 1.1.3 has been installed!
Library Manager: Adafruit GFX Library @ 1.10.11 is already installed
Library Manager: Installing Adafruit TouchScreen
Library Manager: Adafruit TouchScreen @ 1.1.2 has been installed!
Library Manager: Installing Adafruit ILI9341
Library Manager: Warning! More than one package has been found by Adafruit ILI9341 requirements:
 - adafruit/Adafruit ILI9341 @ 1.5.9
 - links2004/Adafruit ILI9341 @ 1.0.0
 - peterus/Adafruit ILI9341 @ 1.5.6
Library Manager: Please specify detailed REQUIREMENTS using package owner and version (showed above) to avoid name conflicts
Library Manager: Adafruit ILI9341 @ 1.5.9 has been installed!
Library Manager: Installing dependencies...
Library Manager: Adafruit GFX Library @ 1.10.11 is already installed
Library Manager: Installing Adafruit SSD1351 library
[... 92 lines cut ...]
**********************************************************************
* Looking for Adafruit_EPD.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:Adafruit_EPD.h"
* Web  > https://platformio.org/lib/search?query=header:Adafruit_EPD.h
*
**********************************************************************

compilation terminated.
Compiling .pio\build\featheresp32\FrameworkArduino\Esp.cpp.o
*** [.pio\build\featheresp32\libd32\Adafruit ImageReader Library\Adafruit_ImageReader_EPD.cpp.o] 
Error 1
.pio\libdeps\featheresp32\Adafruit SPIFlash\src\esp32\Adafruit_FlashTransport_ESP32.cpp: In member function 'SPIFlash_Device_t* Adafruit_FlashTransport_ESP32::getFlashDevice()':
.pio\libdeps\featheresp32\Adafruit SPIFlash\src\esp32\Adafruit_FlashTransport_ESP32.cpp:54:3: error: 'esp_flash_t' was not declared in this scope
   esp_flash_t const *flash = _partition->flash_chip;
   ^
.pio\libdeps\featheresp32\Adafruit SPIFlash\src\esp32\Adafruit_FlashTransport_ESP32.cpp:55:36: error: 'flash' was not declared in this scope
   _flash_device.manufacturer_id = (flash->chip_id >> 16);
                                    ^
*** [.pio\build\featheresp32\lib169\Adafruit SPIFlash\esp32\Adafruit_FlashTransport_ESP32.cpp.o] 
Error 1
================================== [FAILED] Took 37.41 seconds ==================================The terminal process "C:\Users\user\.platformio\penv\Scripts\platformio.exe 'run', '--target', 'upload', '--environment', 'featheresp32'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.
`

hathach, 2021-09-11:
Please edit your post to have full log as attached txt file for readibility

orenzav, 2021-09-11:
Sorry, edited now and attached. Thanks.

hathach, 2021-09-11:
no problem, I will try to reproduce the issue and post more update whenever I could

hathach, 2021-09-13:
@orenzav I don't use platformio, can you post the compile log of the Arduino IDE

orenzav, 2021-09-13:
OK thanks for your help, appreciated

closed 2021-09-13 by orenzav