adafruit/Adafruit_ImageReader #52: [Question] Would a PR Implementing a QOI image decoder be merged?

shraiwi, 2022-04-20:
Hi, I was considering adding a decoder for the [QOI format](https://qoiformat.org/) to this library, to add support for the extremely simple lossless image compression image format. I have [a simple decoder](https://github.com/shraiwi/mini-qoi) that has a memory footprint of about 300 bytes and runs perfectly on an UNO, so do you think that something like that would make sense for this repository, or would it make more sense to keep the QOI decoder as a separate Arduino library? Thanks!

dhalbert, 2022-04-22:
I see this is a brand new format. Some discussion:
https://news.ycombinator.com/item?id=29661498
https://www.reddit.com/r/C_Programming/comments/r2d6wc/qoi_the_quite_ok_image_format_for_fast_lossless/
https://www.reddit.com/r/hackernews/comments/rn6hih/qoi_the_quite_ok_image_format_for_fast_lossless/

shraiwi, 2022-04-22:
Yep! Would it make sense to add support for it in this library?