adafruit/Adafruit_ImageReader #62: RP2040 microSD fails

mikeysklar, 2025-06-27:
I noticed an issue with RP2040 and microSD being a little off while testing the current main branch code yesterday. 

Coming back to it today this is the problem.

Using the [HX8357 3.5" Featherwing TFT](https://www.adafruit.com/product/3651). 

Running the [FeatherWingHX8357 example](https://raw.githubusercontent.com/adafruit/Adafruit_ImageReader/refs/heads/master/examples/FeatherWingHX8357/FeatherWingHX8357.ino).

I'm able to access the microSD using two other boards, but not with a Feather RP2040.

RP2040 error (using 2.9.2 Adafruit_ImageReader release)
```
Initializing filesystem...OK!
Loading adabot.bmp to screen...File not found.
Querying parrot.bmp image size...File not found.
Loading wales.bmp to canvas...File not found.
```

RP2040 error (using current main branch):
```
Initializing filesystem...SD begin() failed
```

Same sketch / display / cord / microSD work fine on:

```
Board                     | 8 MHz | 12 MHz | 25 MHz
--------------------------|-------|--------|-------
Adafruit Feather ESP32 V2 |   ✓   |   ✓    |   ✓
Adafruit Feather M0 Proto |   ✓   |   ✓    |   x
```

mikeysklar, 2025-06-27:
@BlitzCityDIY

mikeysklar, 2025-06-27:
software setup:

```
Component                        | Version       
---------------------------------|---------------
Operating System (macOS)         | Sequoia 15.5  
Arduino IDE                      | 2.3.6         
Adafruit_ImageReader (master)    | master branch 
Adafruit_GFX_Library             | 1.12.1        
Adafruit_BusIO                   | 1.17.1        
Wire                             | 1.0           
SPI                              | 1.0           
Adafruit_HX8357_Library          | 1.1.20        
SdFat – Adafruit Fork            | 2.3.53        
Adafruit_SPIFlash                | 5.1.1         
```

BlitzCityDIY, 2025-06-27:
testing with same hardware and seeing the same issue. i tried the sdinfo example from the sdfat fork and got this output:

```
SdFat version: 2.3.0

Assuming the SD is the only SPI device.
Edit DISABLE_CS_PIN to disable an SPI device.

Assuming the SD chip select pin is: 5
Edit SD_CS_PIN to change the SD chip select pin.

type any character to start

SD initialization failed.
Do not reformat the card!
Is the card correctly inserted?
Is there a wiring/soldering problem?
Is SD_CS_PIN set to the correct value?
Does another SPI device need to be disabled?
SD errorCode: SD_CARD_ERROR_CMD0 = 0x1
SD errorData = 0x0
```

BlitzCityDIY, 2025-06-27:
@mikeysklar i switched to a feather rp2040 prop-maker feather just to double check and the code was working which seemed odd. this reminded me that the rp2040 feather pins are different in arduino, the usual pin 5 is *actually* pin 7. updated the code and it is working

i'll add an ifdef to the example

mikeysklar, 2025-06-27:
@BlitzCityDIY nice work! 

That makes more sense that the RP2040 SD support was never put into the example code. I went on a tangent looking for what had changed.

Tested your ifdef for RP2040 pin7 PR and it worked for me.

```
Initializing filesystem...OK!
Loading adabot.bmp to screen...Success!
Querying parrot.bmp image size...Success!
Image dimensions: 128x160
Loading wales.bmp to canvas...Success!
```

closed 2025-06-27 by mikeysklar